gickowtf/enpal-homeassistant #4: Energy Dashboard

sebastianhi97, 2023-11-30:
Hi,

i got a problem with the entities in the energy dashboard.
[image]
Maybe you know what the problem is. This spike is everyday at the morning.

It seems that the counter didnt reset right, but in the history is the rest at 23:59:35...

valentinbosch, 2024-02-24:
HI @sebastianhi97 ich habe das auch gestern dem HA hinzugefügt und natürlich fleißig das Energie Dashboard versucht zu konfigurieren, sah bei mir heute morgen natürlich genau so aus wie bei dir. Ich vermute, dass man von der Enpal Box keine anständige Entität bekommt, die man direkt im HA Energie Dashboard hinterlegen kann. 

Wie sieht es denn bei dir mittlerweile aus? 

HI @sebastianhi97 I also added this to the HA yesterday and of course diligently tried to configure the energy dashboard, of course it looked exactly like yours this morning. I suspect that you can't get a decent entity from the Enpal Box that you can store directly in the HA Energy Dashboard. 

How are things looking for you now?

gickowtf, 2024-03-29:
hey i tried to fix it by setting all values to 0 in the time from 23:55 to 00:05 ...in version 0.2.1 it is a prerelease

kuroimori, 2024-04-02:
Hi gickowtf,
Thank you for a great HA plugin!
About the midnight reset issue: Have you tried changing the state class for the energy sensors (in kWh/Wh) to 'total_increasing' from 'total' (see e.g. here: https://developers.home-assistant.io/blog/2021/09/20/state_class_total/)?
I tried it locally (lines 182 & 185 in sensor.py) and it's looking fine for me.

gickowtf, 2024-04-02:
thanks for the info i will change it right away and push it

b-huessner, 2024-04-04:
That has fixed it for me, thank you

gickowtf, 2024-04-05:
Yes, the Update have fixed the issue.