gickowtf/enpal-homeassistant #9: Can't get values of the FOXess Battery nor Inverter?

GetUpdate, 2024-03-06:
First at all, THE for the integration :-)

Today my ENPAL Solar system had been connected.
FOX Ess Inverter and Battery plus ENPAL Box.
The Inverter is connected via LAN cable with the Enpal Box. The Enpal Box is via LAN cable connected to my Switch.
The Enpal Box is -up to now- in the same LAN (would like to put it later in to separate vLAN).

I could access the Box as explained in the Read.me, but I only get the one of the three cards (Enpal Energy Consumption , DAY, TOTAL...), I currently don't get the Battery nor the Inverter.

Kindly ask for your support / hints please

M

SvenAbels, 2024-03-29:
I got the same problem here with my new fox ESS.  I'm trying to integrate your changes but I'm not sure where.   Do I understand you right, that you modified the original sensor.py file from the extension by simply editing the file from the custom_components\enpal\ folder?

@GetUpdate    Maybe you can attach your sensor.py file here?

@gickowtf  Of course it would also be fantactic if it could be added to the official extension

SvenAbels, 2024-03-29:
@GetUpdate I managed to integrate it succesfully.  But I believe that the value with the String 1 and Sring 2 is wrong in your script above: They do not represent the phases but the Watt amount of the 2 PV strings connected to the Enpal system

gickowtf, 2024-03-29:
i have published a prerelease v0.2.1 https://github.com/gickowtf/enpal-homeassistant/releases/tag/0.2.1

gickowtf, 2024-03-29:
hey ich glaube für können da es um enpal geht auch deutsch schreiben ;D ich habe gerade eben mal ein pre release rein gestellt... habe ihr Zugang zu influxdb?? könnt ihr mir die genauen `fields` nennen. dann kann ich nochmal ein vernünftiges update machen.

SvenAbels, 2024-03-29:
> hey ich glaube wir können, da es um enpal geht, auch deutsch schreiben ;D ich habe gerade eben mal ein pre release rein gestellt... 

Klasse, vielen Dank. :-)

> habe ihr Zugang zu influxdb?? könnt ihr mir die genauen `fields` nennen. dann kann ich nochmal ein vernünftiges update machen.

Leider habe ich keinen Zugang zur Influxdb.  Ich hab die Konfigurationsbeispiele oben etwas abgeändert, indem ich nicht die Phasen A/B/C, sondern die Strings abgefragt habe und mir ausgeben lasse. Das klappt super:

`        #Power Sensor -> Inverter
        if measurement == "inverter" and field == "Voltage.String.1":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Voltage String 1', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'voltage', 'V'))
        if measurement == "inverter" and field == "Current.String.1":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Ampere String 1', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'current', 'A'))
        if measurement == "inverter" and field == "Power.DC.String.1":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Power String 1', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'power', 'W'))
        if measurement == "inverter" and field == "Voltage.String.2":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Voltage String 2', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'voltage', 'V'))
        if measurement == "inverter" and field == "Current.String.2":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Ampere String 2', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'current', 'A'))
        if measurement == "inverter" and field == "Power.DC.String.2":
            to_add.append(EnpalSensor(field, measurement, 'mdi:lightning-bolt', 'Enpal Power String 2', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'power', 'W'))
`

Ich bekomme aber bei meiner FoxESS Anlage leider keinen Wert für 'Enpal Energy External In Day' (bzw. immer 0 kwh), obwohl für 'Enpal Energy External Out Day' alles timmt. Ich vermute aber, dass das kein Problem deines Skriptes ist, denn in dem Enpal Solar Rel wird der Wert ebenfalls mit 0 aufgeführt (anders als in der App).

Was ich noch nicht verstehe ist die Bedeutung von "Enpal Energy Consumption". Was verbirgt sich dahinter?

Viele Grüße und ein frohes Osterfest :-),

Sven

SvenAbels, 2024-04-02:
Hi,

ein kurzer Nachtrag: Ich habe die `if measurement == "..." ` statements nun mit der Ausnahme vom Power-Abschnitt ganz bei mir rausgenommen, da die Werte andernfalls ab und zu nicht übertragen wurden.  Nun läuft alles stabil mit der FoxESS - bzw. zumindest mit den Werten, die ich auch im Solar Rel sehe.

skehrer, 2024-04-02:
> Klasse, vielen Dank. :-)
>
>
> Leider habe ich keinen Zugang zur Influxdb. Ich hab die Konfigurationsbeispiele oben etwas abgeändert, indem ich nicht die Phasen A/B/C, sondern die Strings abgefragt habe [...]

Hi Sven,

Ich  habe in meiner Übersicht auch meistens für die "Enpal Energy Consumption" 0kWh stehen. Aktuell steht der Wert auf 0,5kWh. So wie ich das verstehe ist das der Gesamtbezug deines Hauses aus dem Netz. Wenn Du also ein System mit Speicher hast der über die Nacht reicht und Du momentan mehr produzierst als Du verbrauchst macht es Sinn, dass da eine 0 steht. Das ist bei mir der Fall, es schient aber so als ob über den Tag verteilt eine gewisse Minimalmenge immer aus dem Netz bezogen wird. Ich habe seit der Installation der Anlage vor ca 2 Wochen im Schnitt einen täglichen Bezug aus dem Netz con ca 0,5kWh gehabt.

Gruß,
Stephan

SvenAbels, 2024-04-02:
Hi Stephan,

ich habe noch einen zweiten Smart Meter von der Wallbox und auch den Zähler etwas beobachtet. Es scheint so als wenn der Gesamtverbrauch aus dem Netz immer "Enpal Energy Consumption" + " Energy External In Day" entspricht. Bei mir sind beispielsweise heute beide Werte gefüllt - einmal mit 1,1kwh und einmal mit 0,2kwh. Der korrekte Wert lauf meinem SmartMeter der Wallbox liegt bei 1.3...

skehrer, 2024-04-02:
> hey ich glaube wir können, da es um enpal geht, auch deutsch schreiben ;D ich habe gerade eben mal ein pre release rein gestellt... habe ihr Zugang zu influxdb?? könnt ihr mir die genauen `fields` nennen. dann kann ich nochmal ein vernünftiges update machen.

Hi Gicko,

Ich habe seit knapp 2 Wochen ebenfalls ein System von Enpal mit Inverter und 20 kWh Speicher vonFox ESS. Heute habe ich auch den "viewer"-Zugang von Enpal erhalten und mir Deine Integration gleich mal installiert und angeschaut. Vielen Dank für die Arbeit die Du da investiert hast.

Bei mir funktionieren ebenfalls momentan einige der Sensoren nicht. Ich sehe mit der aktuellen Version der Integration folgendes:
- Enpal Energy Consumption
- Enpal Energy External In Day
- Enpal Energy External Out Day
- Enpal Power External Total
- Enpal Power House Total
- Enpal Production Day
- Enpal Solar Production Power

Was mir nicht angezeigt wird sind sämtliche Batterie bezogenen Werte sowie die Enpal Voltage Phase A-C, Enpal Power Phase A-C, Enpal Ampere Phase A-C und die Werte zur Wallbox (habe ich keine).

Auf die Schnelle musste ich feststellen, dass bei mir die relevanten measurements teilweise anders heißen als in Deiner Integration. Ich habe zum Beispiel kein measurement namens "battery". Bei mir stehen die entsprechenden Felder zum Großteil im measurement "inverter" und (teilweise gedoppelt und mit anderem Feldbezeichner) im measurement "system".

Ein schneller Test bei dem ich das measurement für die Batteriegruppe durch "inverter" ersetzt habe hat dafür gesorgt, dass ich jetzt die entsprechenden Batteriesensoren angezeigt bekomme. Ich werde mir das in den nächsten Tagen mal genauer anschauen und Dir dann meinen Fix zukommen lassen. Ich kann Dir auch gerne das genaue Schema der influxDB die auf meinem System läuft (Solar Rel 8.35) zukommen lassen wenn Du mir sagst, auf welchem Weg ich das am besten machen soll.

Als weitere kurze Information habe ich vom Enpal Service-Team noch folgende Aussage bekommen: "Wir arbeiten in diesem Jahr an einer größeren Umstrukturierung unserer Lösung die dann auch auf Ihrem System ausgerollt wird. Dies wird dann auf
jeden Fall auch die Influx DB betreffen. Sie müssten dann noch einmal Anpassungen an Ihrem System vornehmen."

Nur als kleine Warnung vorab, dass zumindest bei den neuen Enpal-Boxen wohl in absehbarer Zukunft noch das eine oder andere an Anpassungen fällig werden wird.

Gruß,
Stephan

GetUpdate, 2024-04-07:
> Hi Gicko,
>
> Ich habe seit knapp 2 Wochen ebenfalls ein System von Enpal mit Inverter und 20 kWh Speicher vonFox ESS. Heute habe ich auch den "viewer"-Zugang von Enpal erhalten und mir Deine [...]

Hallo, 
Sorry für die späte Rückmeldung, war 2 Wochen am Roten Meer und leider ist dort kein VPN möglich... 
Sehe auch wie @skehrer nur die Enitäten wie oben beschrieben. 

Mir fehlt mit V0.2.2 noch immer die Werte wie ich oben angegeben habe. 
Mit meiner eigenen Sensor.py klappt das jedoch wieder. 

@skehrer kannst Du ja mal versuchen, Die Datei findest Du unter 
/homeassistant/custom_components/enpal/sensor.py 
-> Umbenennen und als Kopie behalten
-> Inhalt von sensor.py mit werten unten überschreiben. 
-> HA neu starten...

Gruß Michael 

Sensor.py:

"""Platform for sensor integration."""
from __future__ import annotations

import asyncio
import uuid
from datetime import timedelta, datetime
from homeassistant.components.sensor import (SensorEntity)
from homeassistant.core import HomeAssistant
from homeassistant import config_entries
from homeassistant.helpers.device_registry import DeviceEntryType
from homeassistant.helpers.entity import DeviceInfo
from homeassistant.helpers.entity_registry import async_get, async_entries_for_config_entry
from custom_components.enpal.const import DOMAIN
import aiohttp
import logging
from influxdb_client import InfluxDBClient

_LOGGER = logging.getLogger(__name__)
SCAN_INTERVAL = timedelta(seconds=20)

VERSION= '0.1.0'

def get_tables(ip: str, port: int, token: str):
    client = InfluxDBClient(url=f'http://{ip}:{port}', token=token, org='enpal')
    query_api = client.query_api()

    query = 'from(bucket: "solar") \
      |> range(start: -5m) \
      |> last()'

    tables = query_api.query(query)
    return tables
 

async def async_setup_entry(
    hass: HomeAssistant,
    config_entry: config_entries.ConfigEntry,
    async_add_entities,
):
    # Get the config entry for the integration
    config = hass.data[DOMAIN][config_entry.entry_id]
    if config_entry.options:
        config.update(config_entry.options)
    to_add = []
    if not 'enpal_host_ip' in config:
        _LOGGER.error("No enpal_host_ip in config entry")
        return
    if not 'enpal_host_port' in config:
        _LOGGER.error("No enpal_host_port in config entry")
[... 125 lines cut ...]
                    self._attr_icon = "mdi:battery-10"
                if self._attr_native_value <= 29 and self._attr_native_value >= 20:
                    self._attr_icon = "mdi:battery-20"
                if self._attr_native_value <= 39 and self._attr_native_value >= 30:
                    self._attr_icon = "mdi:battery-30"
                if self._attr_native_value <= 49 and self._attr_native_value >= 40:
                    self._attr_icon = "mdi:battery-40"
                if self._attr_native_value <= 59 and self._attr_native_value >= 50:
                    self._attr_icon = "mdi:battery-50"    
                if self._attr_native_value <= 69 and self._attr_native_value >= 60:
                    self._attr_icon = "mdi:battery-60"
                if self._attr_native_value <= 79 and self._attr_native_value >= 70:
                    self._attr_icon = "mdi:battery-70"
                if self._attr_native_value <= 89 and self._attr_native_value >= 80:
                    self._attr_icon = "mdi:battery-80"
                if self._attr_native_value <= 99 and self._attr_native_value >= 90:
                    self._attr_icon = "mdi:battery-90"        
                if self._attr_native_value == 100:
                    self._attr_icon = "mdi:battery"
                
        except Exception as e:
            _LOGGER.error(f'{e}')
            self._state = 'Error'
            self._attr_native_value = None
            self._attr_extra_state_attributes['last_check'] = datetime.now()

skehrer, 2024-04-08:
Hi Gicko,

Danke für die angepasste Sensor.py. Ich teste morgen (sofern ich dazu komme) mal aus ob das bei mir funktioniert. Ich habe mir aber die statements gerade mal angeschaut und denke das müsste passen. Gibt es einen Grund warum Du bei den Sensoren für die Batterie das measurement aus der if-Abfrage rausgenommen hast? Ich hätte eher mit sowas gerechnet:
```
        if measurement == "inverter" and field == "Power.Battery.Charge.Discharge":
            to_add.append(EnpalSensor(field, measurement, 'mdi:battery-charging', 'Enpal Battery Power', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'power', 'W'))
```

Was ich heute abend mal gemacht habe ist mir den aktuellen Aufbau des solar-Buckets anzuschauen. In der aktuellen Version (immer noch Solar Rel 8.35 vom 25.03.2024) habe ich im measurement "inverter" etliche zusätzliche Felder gegenüber dem letzten Mal als ich mir das angeschaut habe. Das Schema sieht bei mir aktuell so aus:

Measurement "System":
```
Energy.Consumption.Total.Day
Energy.External.Total.In.Day
Energy.External.Total.Out.Day
Energy.Production.Total.Day
Energy.Storage.Level
Energy.Storage.Total.In.Day
Energy.Storage.Total.Out.Day
Percent.Storage.Level
Power.Consumption.Total
Power.External.Total
Power.Production.Total
Power.Storage.Total
measureId
```

Measurement "inverter":
```
Battery.ChargeLevel.Max
Battery.ChargeLevel.Min
Battery.ChargeLevel.MinOnGrid
Battery.SOH
Command.Start.Stop
Current.Battery
Current.String.1
Current.String.2
Dongle.BackendConnection
Energy.Battery.Ah
Energy.Battery.Charge.Day
Energy.Battery.Charge.Level
Energy.Battery.Charge.Load
Energy.Battery.Discharge.Day
Energy.Consumption.Total.Day
Energy.Grid.Export.Day
Energy.Grid.Import.Day
Energy.Production.Total.Day
Frequency.Grid
Inverter.System.State
Meter.Connect.State
Meter.Enable.Disable
Meter2.AC.Phase.A
Meter2.AC.Phase.B
Meter2.AC.Phase.C
Meter2.AC.Total
Meter2.Connect.State
Meter2.Enable.Disable
Mode.Forcible.Charge.Discharge
Mode.Forcible.Timeout
Power.AC.Phase.A
Power.AC.Phase.B
Power.AC.Phase.C
Power.Battery.Charge.Discharge
Power.Battery.Charge.Discharge.Set
Power.DC.String.1
Power.DC.String.2
Power.DC.Total
Power.Factor
Power.Grid.Export
Power.House.Total
Power.Limit
Power.Reactive
Voltage.Battery
Voltage.Phase.A
Voltage.Phase.B
Voltage.Phase.C
Voltage.String.1
Voltage.String.2
```

Measurement "iot":
```
Cpu.Load
LTE.Predictor.Result.Passed
LTE.Quality
LTE.RSRP
LTE.RSRQ
LTE.RSSI
LTE.SNR
Memory.Usage
```

Measurement "wallbox":
```
Power.Wallbox.Connector.0.Charging.Requested
Power.Wallbox.Connector.1.Charging.Requested
Status.Wallbox.Connected
```

Einige Felder scheinen dabei Dopplungen zu sein, z.B. "system-Energy.Storage.Total.In.Day" und "inverter-Energy.Battery.Charge.Day", die beide "Enpal Battery Charge Day" anzuzeigen scheinen.

Ich nehme mir in den nächsten Tagen mal Deine originale Sensor.py und passe sie so an, dass sie für mein System passt. Wenn ich das gemacht habe stelle ich sie Dir hier in einen Post.

Gruß,
Stephan

skehrer, 2024-04-11:
Interessant, dass das bei Dir voneinander abweicht. Ich habe gerade eben mal geschaut und zumindest stand heute abend 22 Uhr sind die Werte unter System und Inverter bei mir identisch. Ich schaue mir das morgen früh nochmal an. Mal schauen ob es da bei mir dann trotzdem identische Werte sind oder ob das einfach erst im Lauf des Tages nachgezogen wird.

Bezüglich der Uhrzeit: nein, ich habe keine PIN für die Inverter Config. Welche Uhrzeit meinst Du, die bei Dir nicht korrekt ist? Bei mir ist die Uhrzeit der Werte im Menüpunkt "Messages" ebenfalls um zwei Stunden daneben. Allerdings gehe ich davon aus, dass diese Uhrzeit in UTC ausgeliefert wird. Das würde passen da es vor der Zeitumstellung bei mir eine Stunde war. Die Zeitstempel in der InfluxDB sind bei mir korrekt in CEST. Das scheint mir also (auf meinem System) soweit alles zu passen.

Kleine Randbemerkung noch: ich habe gerade gesehen dass ich seit heute eine neue Version der Solar habe, nämlich Rel 8.35.1 vom 10.04.2024. Das Schema der InfluxDB ist aber unverändert!

SvenAbels, 2024-04-11:
> Hat eigentlich einer von euch die Enpal PIN für die Inverter Config? Leider hat sich bei mir die Sommerzeit noch nicht umgestellt :(

Das habe ich hier auch und das Zeitproblem führte bei mit dazu, dass die Werte von Energy.Production.Total.Day z.b. verspätet auf 0 gesetzt wurden und daurch bei HomeAssistant das Energiedashboard nicht mehr stimmte. Ich habe das dadurch gelöst, dass ich einen Verbrauchszähler erstellt habe, der die Werte immer aufsummiert und dass ich den dann als Quelle im Energiedashboard genommen habe...

TheFlea80, 2024-04-12:
> Bezüglich der Uhrzeit: nein, ich habe keine PIN für die Inverter Config. Welche Uhrzeit meinst Du, die bei Dir nicht korrekt ist? Bei mir ist die Uhrzeit der Werte im Menüpunkt "Messages" ebenfalls [...]

Bei mir passt die Uhrzeit auf dem Inverter selbst nicht. Die Daten in der Datenbank sind mit dem korrekten UTC-Zeitstempel versehen. Es ist mehr ein "Es triggert mich" als ein technisches Problem.

> Kleine Randbemerkung noch: ich habe gerade gesehen dass ich seit heute eine neue Version der Solar habe, nämlich Rel 8.35.1 vom 10.04.2024. Das Schema der InfluxDB ist aber unverändert!

Ja, das kann ich bestätigen.

Was mir auch noch aufgefallen ist: 
Der Wert "Energy.External.Total.In.Day" (aktuell 6,6kWh) und "Energy.Grid.Import.Day" (aktuell 1,1kWh) sind beide unterschiedlich bei mir. Und beide sind falsch! In der ENPAL App allerdings wird der richtige Wert angezeigt (aktuell 3,4kWh). Ich verstehe zum verrecken nicht, wo die App sich den Wert herzieht.
Soweit ich es rekonstruieren kann, ist der Wert Energy.External.Total.In.Day der gestrige Verbrauch. Bei den anderen Werte, stehe ich auf dem Schlauch...

SvenAbels, 2024-04-12:
> Was mir auch noch aufgefallen ist: Der Wert "Energy.External.Total.In.Day" (aktuell 6,6kWh) und "Energy.Grid.Import.Day" (aktuell 1,1kWh) sind beide unterschiedlich bei mir. Und beide sind falsch! [...]

Schau mal, ob Energy.Grid.Import.Day + Energy.Consumption.Total.Day in Summe bei dir die 3,4 kWh ergeben. Bei mir ist das so, dass beide Werte zusammen dem Netzimport entsprechen

TheFlea80, 2024-04-12:
> Schau mal, ob Energy.Grid.Import.Day + Energy.Consumption.Total.Day in Summe bei dir die 3,4 kWh ergeben. Bei mir ist das so, dass beide Werte zusammen dem Netzimport entsprechen

Leider nein. Der Energy.Consumption.Total.Day Wert, ist bei mir der Gesamtverbrauch des Hauses.
Ich müsste also, um mir den Wert zu errechnen folgendes tun:
Gesamtverbrauch -Batterieladen -Batterieentladen -solarerzeugungTag -netzeinspeisung rechnen. 
Alternativ baue ich mir einen ESP der mir den Verbrauch am Zähler direkt ermittelt. Das System von ENPAL scheint noch nicht richtig ausgereift zu sein...

GetUpdate, 2024-04-15:
Denke vom Zähler, den siehst du in influx DB nicht…Von meinem iPhone gesendet

GetUpdate, 2024-04-29:
Hallo Zusammen, 

habe dieses issue gestartet um die Fox Ess Harware ebenfalls auslesen zu können. 
Leider geht das mit der standard integration nach wie vor nicht.  Nutze daher noch immer eine angepasste sensor.py. 
Ist es möglich diese zu integrieren oder braucht man für Fox Ess eine eigene Integration?

Gruß und Dank im Voraus 
M

UniCizin, 2024-06-14:
habs probiert und funtkioniert einwandfrei.
Habe dann jetzt auch 17 Entitäten, Werte selbst habe ich noch nicht gecheckt

JuergenNe, 2024-06-24:
Hallo, ich bekomme die Entität Batterie nicht. In der Enpal Konfiguration bzw. im Data-Collektor ist sie mit Storage benannt.
Kennt jemand von euch das Problem. Leider bin ich noch ein ziemlicher Neuling und bekomme die Änderung alleine auch nicht hin.

JuergenNe, 2024-06-25:
Hallo, ich habe meine Anlage mit den Komponenten von Fox vor 1 Woche bekommen. Es scheint, dass Enpal die Datenstruktur schon wieder ändert hat. Ich habe mit der Mail von Enpal mit den Zugangsdaten auch die Information bekommen, dass:

"Die InfluxDB ist auch Teil unserer zentralen SW-Lösung. Falls notwendig, kann es sein, dass wir hier kurzfristig Anpassungen durchführen müssen.
Wir arbeiten in diesem Jahr an einer größeren Umstrukturierung unserer Lösung, die dann auch auf Ihrem System ausgerollt wird. Dies wird dann auf
jeden Fall auch die InfluxDB betreffen. Sie müssten dann noch einmal Anpassungen an Ihrem System vornehmen."

Ich kann gerne versuchen die Datenbankstruktur auszulesen. Vielleicht kann mir dann jemand helfen, die Änderungen vorzunehmen.

GetUpdate, 2024-06-26:
Enpal hat mir mitgeteilt das weitere Änderungen anstehen. 
Können wir uns also eh drauf einstellen. 

Gruß

SvenAbels, 2024-06-26:
> Enpal hat mir mitgeteilt das weitere Änderungen anstehen. Können wir uns also eh drauf einstellen.

Sollten sie zu viel umbauen, könnten wir zur Not auch die Webseite der Enpal-Box parsen. Dafür habe ich mir ein kleines Skript geschrieben. Das ist zwar nicht so schön wie das Abfragen der InfluxDB, aber es erfüllt seinen Zweck...

Viele Grüße,

Sven

GetUpdate, 2024-06-30:
> Sollten sie zu viel umbauen, könnten wir zur Not auch die Webseite der Enpal-Box parsen. Dafür habe ich mir ein kleines Skript geschrieben. Das ist zwar nicht so schön wie das Abfragen der InfluxDB, [...]

Cool :-)  Gut zu Wissen.  Gruß

skehrer, 2024-06-30:
Ich habe mir gerade mal das aktuelle DB Format (bei mir die 8.39.1 vom 21.06.2024) angeschaut. So viel hat sich zur Version 8.35.1 tatsächlich gar nicht getan. Im Measurement "inverter" sind ein paar Werte dazugekommen, aber die bereits bestehenden Werte wurden nicht verändert. Neu sind:

- Energy.Battery.Charge.Level.Absolute
- SoftwareVersion.Service.Battery.Unit.1
- State.ErrorCodes.1
- State.ErrorCodes.10
- State.ErrorCodes.11
- State.ErrorCodes.2
- State.ErrorCodes.6
- State.ErrorCodes.9
- Temperature.Battery

In den anderen Measurements gab es keine Veränderungen.

Meine für die 8.35.1 lokal angepasste Version der sensor.py liefert mir nach wie vor alle relevanten Werte. Ich habe meine aktuelle Version der sensor.py mal an diesen Post angehängt.
@JuergenNe Du kannst ja mal testen ob Du mit der version auch die Werte für die Batterie siehst. Bei mir klappt es und ich habe auch die FoxESS Komponenten. Github hat mich hier leider keine .py Datei anhängen lassen, daher habe ich sie zu .txt umbenannt. Du musst die Datei also wieder von einer .txt. zu einer .py umbenennen. 

Gruß,
Stephan
[sensor.txt](https://github.com/user-attachments/files/16045943/sensor.txt)

JuergenNe, 2024-07-01:
@skehrer 
Hallo, jetzt hab ich auch 22 Entitäten. Super. Vielen Dank.

Darf ich unverschämt ragen, ob du ein ansprechendes Dashboard schon konfiguriert hast, das ich verwenden könnte.

Vielen Dank

Gruß,
 Jürgen