gickowtf/enpal-homeassistant #10: Enpal Energy Consumption

SvenAbels, 2024-03-29:
Hi,

I do not understand what the Enpal Energy Consumption could be.  Does anyone have an explanation what this value means?

b-huessner, 2024-04-02:
Hi,

I think that is your house's energy consumption.

SvenAbels, 2024-04-02:
Yes, that is what I tought as well but the values don't add up - at least with my device (FoxESS). However, it seems to be that the sum of "Enpal Energy Consumption" + " Energy External In Day" is identical to the energy consumed from the grid.   At least they add up to the same value that my other smart meter and that the measurement meter of the grid provider show.

This is strange especially because "Energy External Out Day" is exactly identical to the energy provided to the grid.

skehrer, 2024-04-03:
It used to show the daily energy consumption for the house for me. However, since last night the value has changed. I now only see a very gradual increase and it shows a total of 0.5kWh now, at 22:30. The enpal app shows a total consumption of 9.7kWh, which looks more plausible to me. Has the value changed to something unreasonably low for anybody else, i.e. did some logic in the box change, or is this only me?

I also can calculate the total consumption of the house (matching exactly with what the app shows me) my doing the following calculation:
Total consumption house = Enpal Production Day - Enpal Energy External Out Day - Enpal Battery Charge Day + Enpal Battery Discharge Day +  Enpal Energy Consumption
But as I said, the value is not showing up directly in Enpal Energy Consumption any longer, as ist used to do up until yesterday evening...