gickowtf/enpal-homeassistant #12: Repository not available anymore in Hacs

Busterdick, 2024-07-11:
Tried to install it but it is nit available in HACS anymore. If installed manually, HA crashes fully (2024.7.2). Needed to re install HA and use a backup…..
Anyone an idea ??

GetUpdate, 2024-07-12:
Hi Busterdick, 

could find the Integration via HACS 
-> Sorry, are you shuraf that you search via HACS Integration?
Hint: did the latest core update yesterday.
Kind Regards

mcpdev80, 2024-08-27:
Hi I can't find it in Hacs or in integration and if I try to use the link I get:

Error: This integration does not support configuration via the user interface. If you followed this link from the Home Assistant website, please ensure that you are running the latest version of Home Assistant.

My HA is up-to-date, and Hacs also?

michasoellner, 2024-11-02:
Same Probleme here.

fabfive23, 2025-01-21:
same problem here
anyone fixed it?