gickowtf/enpal-homeassistant #13: Battery charge/discharge data only in full KWH in Rolar Rel 8.40.1 (23.07.2024)

SvenAbels, 2024-08-03:
Hi,

I noticed that the battery charge/discharge data is only available as an int data in full KHW values from the latest Rolar Rel 8.40.1 (23.07.2024).  For exmaple, before the charge/discharge was showing e.g. 0,2kwh battery charge. Now, it stays 0 until 1 kwh is achieved.  This leads to strange displays at the home assistant dashboard.

I got a FoxESS battery. I can also see the same problem on the enpay web UI at http://enpal/deviceMessages

Can anyone confirm this problem

tankanickel, 2024-08-08:
I can confirm this. 
The total battery discharge amount has also vanished since quite some time.

SvenAbels, 2024-08-08:
The problem seems to also concern the KWH of the power production.
I was, however, able to create a manual riemann integral calculation template for calculating the values based on the current W numbers. This worked but it only a temporar workaround

tankanickel, 2024-08-08:
The power production is a W value on my end. Only the "Day" values are kWh.
And now I see the actual thingy -> The momentary values are all in Watts, while the Daily values are only counted as soon as a full kW was reached.

SvenAbels, 2024-08-08:
Yes. What I mean is that you can create a template with a Riemann Integral Sum to sum up the W values over time to KWh values