gickowtf/enpal-homeassistant #16: Access Token for InfluxDB

chrisniegisch64, 2024-09-03:
Dear all,
I am a bit confused about how to get the Access Token for the Enpal Integration. I managed to add the Rep and the Integration but I am really not sure where to get the Access Token from.
Beeing quite an Newbie with Homeassistant I will be thankful to push me in the right direction.
Many thanks in advance!
Chris

tankanickel, 2024-09-06:
Hi,
You need to get the influxdb information out of the webinterface of the inverter (the Solar Rel. .... thingy). Open the ip up, go to Config and scroll down on the bottom until you get the influx:Token.

chrisniegisch64, 2024-09-06:
Many thanks,found it already!Mit freundlichen Grüßen Christopher ***@***.***: +49 170 7391168Festnetz: + 49 6182 9927178Fax: + 49 6182 9927179 Facebook: Just the Two of US - https://www.facebook.com/Just-the-Two-of-Us-252504752437418

AntiVirus & Spam checked by Sophos Antivirus

tommy0code, 2024-09-27:
Where can I find that?

I have this in my config:

`{
  "ModbusDevices": {
    "0": {
      "NodeAddress": "1"
    },
    "1": {
      "NodeAddress": "1"
    }
  },
  "MainState": "Operation",
  "IotHubClient": {
    "ConnectionString": "..."
  },
  "MqttSettings": {
    "Username": "...",
    "Password": "..."
  },
  "ModbusTcpInterfaceSetting": {
    "0": {
      "IpAddress": "192.168.2.133"
    }
  }
}`

But I can reach the influx db with 192.168.2.125:8086 
influxdb version 2.2.0

tommy0code, 2024-09-27:
found it myself now.
Its not in the config itself its at the bottom of the page