gickowtf/enpal-homeassistant #21: No longer receiving battery metrics

CamBanfield, 2024-11-24:
I'm not using this integration, however I'm pretty sure you will be facing the same issue.

As of the latest Enpal box update `Solar Rel. 8.43.1+243312 (18.11.2024)` it appears that a few metrics are no longer being saved into influxdb.

- All battery metrics are now missing
- House total power consumption has been renamed to `Power.Consumption.Total` under `system` and not `inverter`

I had made a custom Grafana dashboard and I now no longer have metrics for any battery data :( 

How supported is the access to influxdb, could one submit a ticket to Enpal and report the missing metrics? I found access to the InfluxDB and token a while back and didn't know if this is supported or even "allowed" by Enpal. Over the past year I have noticed a lot of values keep moving around, names changing and at some cases broken (the kwh import / export was reversed for a month)

Wondering if anyone else has lost battery metrics along with the latest update?

Many thanks

sceife, 2024-11-24:
I have the same problems since the update inside the enpal box.

MickyPf, 2024-11-24:
I have the same problems since the update inside the enpal box.

michaelhome19, 2024-11-24:
Since 2024/11/20 I lost also all values for Battery metrics. I use HA and Grafana/InfuxDB. I doesn't found any renames for all this Value in Who can fix this Problem??  
[image]

My values for import/export solar also changed from "+" to "-" some times ago!!

michaelhome19, 2024-11-24:
I have fixed the sensor.py (-->/homeassistant/custom_components/enpal/sensor.py) and in sector #battery new measurements and fields changed. Now it works fine with new values!

Please send me an answer if this best way to fix the Problem! 
THX 
Michael

 
**line:66**
        # (Änderung 24.11.2024 ENtitäten werden nicht in der InfuxdB unter diesem measurement zur Verfügung gestellt)
        if measurement == **"system" and field == "Power.Consumption.Total"**:
            to_add.append(EnpalSensor(field, measurement, 'mdi:home-lightning-bolt', 'Enpal Power **Verbrauch** Total', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'power', 'W'))

**line:105**
#Battery (Änderung 24.11.2024 ENtitäten werden nicht in der InfuxdB unter dem measurement battery zur Verfügung gestellt)   
        if measurement == **"system" and field == "Power.Storage.Total":**
            to_add.append(EnpalSensor(field, measurement, 'mdi:battery-charging', 'Enpal Battery Power', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'power', 'W'))
        if measurement == **"system" and field == "Percent.Storage.Level":**
            to_add.append(EnpalSensor(field, measurement, 'mdi:battery', 'Enpal Battery Percent', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'battery', '%'))
        if measurement == **"system" and field == "Energy.Storage.Total.In.Day":**
            to_add.append(EnpalSensor(field, measurement, 'mdi:battery-arrow-up', 'Enpal Battery Charge Day', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'energy', 'kWh'))
        #if measurement == "battery" and field == "Energy.Storage.Total.Out.Day":
        #    to_add.append(EnpalSensor(field, measurement, 'mdi:battery-arrow-down', 'Enpal Battery Discharge Day', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'energy', 'kWh'))
        #if measurement == "battery" and field == "Energy.Battery.Charge.Total.Unit.1":
        #    to_add.append(EnpalSensor(field, measurement, 'mdi:battery-arrow-up', 'Enpal Battery Charge Total', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'energy', 'kWh'))
        #if measurement == "battery" and field == "Energy.Battery.Discharge.Total.Unit.1":
        #    to_add.append(EnpalSensor(field, measurement, 'mdi:battery-arrow-down', 'Enpal Battery Discharge Total', config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'], 'energy', 'kWh'))

Ringo2708, 2024-11-26:
Hallo,
ich habe die sensor.py gemäß der Anleitung gefixt. Nun habe ich zumindest 10 anstatt 6 Etitäten. Mir fehlen aber trotzdem noch 12 Entitäten. Ich hatte ursprünglich mal 22 Entitäten.
Kann da jemand weiterhelfen?
Vielen Dank

Gruß
Ringo

MickyPf, 2024-11-26:
Ich habe diese Zip-Datei aus Isusses entities missing benutzt. Jetzt habe ich 22 Entitäten.  Fast alles Prima. Leider wird im Energie-DashBoard die Entität "Enpal Solar Production Power" bei Solarproduktion nicht angezeigt.

LG
Michael

Ringo2708, 2024-11-27:
@MickyPf 
Danke für den Tipp. Ich habe die sensor.py ausgetauscht und habe nun sogar 25 Entitäten. Muss mich nur nur noch mal damit befassen welche das alle sind.

Gruß
Ringo

HealsCodes, 2024-11-30:
Somewhat reassuring to see others are getting missing metrics as well.

Our box is now running `Solar Rel. 8.43.3+249078 (28.11.2024)` and the influx bucket for `solar` hasn't received any records for almost a week (queries work fine but there's just no data to return).

Anyone with similar experiences?

sebastianhi97, 2024-12-18:
Same here, 
after adjusting the code like u said.
I got my battery status and charching power back. But the House Power is still missing..

TerranUlm, 2025-02-13:
I simply removed all the checks for the measurements as the fields are all unique in my system. Now I get 31 entities in my system 🎱

Dane2909, 2025-02-18:
Hi, I can't even get the battery working again. Could someone upload their entire "sensor.py" here? Thanks

TerranUlm, 2025-02-24:
I've made some changes to the sensor.py, so you get most of the data, even if the sensor names change. The only time it needs to be updated is if there is a new type of measurement which isn't covered in the code so far.
The downside of this is that you might need to make changes to your homeassistant configuration to make use of the changed sensor names but at least you will have them available immediatly.

I also added so **short** explanations on how to use the sensors for the energy dashboard. The battery values are a bit tricky to get the same values as in the enpal app and I couldn't test the "discarge" value so far as there wasn't much sunshine so far...

So check out for yourself: https://github.com/TerranUlm/enpal-homeassistant/tree/TerranUlm-patch-1

Daniel7729, 2025-02-25:
great work!
now i have 57 entitäten :)