gickowtf/enpal-homeassistant #28: issues to connect to Solar Rel. 8.45.1-301788 (10.03.2025)

fluidman, 2025-03-11:
looks like there was an update on 10.03.2025

first i lost the battery entities, and when i tried to remove and readd the integration with my existing Influx-DB Token i now get the below error messages:
Protokolldetails (FEHLER)
Logger: homeassistant.components.sensor
Quelle: helpers/entity_platform.py:382
Integration: Sensor ([Dokumentation](https://www.home-assistant.io/integrations/sensor), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Erstmals aufgetreten: 20:56:35 (4 Vorkommnisse)
Zuletzt protokolliert: 23:12:25

Error while setting up enpal platform for sensor: (401) Reason: Unauthorized HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.2.0', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Tue, 11 Mar 2025 19:56:35 GMT', 'Content-Length': '55'}) HTTP response body: b'{"code":"unauthorized","message":"unauthorized access"}'
Error while setting up enpal platform for sensor: (401) Reason: Unauthorized HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.2.0', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Tue, 11 Mar 2025 19:57:01 GMT', 'Content-Length': '55'}) HTTP response body: b'{"code":"unauthorized","message":"unauthorized access"}'
Error while setting up enpal platform for sensor: (401) Reason: Unauthorized HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.2.0', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Tue, 11 Mar 2025 19:58:07 GMT', 'Content-Length': '55'}) HTTP response body: b'{"code":"unauthorized","message":"unauthorized access"}'
Error while setting up enpal platform for sensor: (401) Reason: Unauthorized HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.2.0', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Tue, 11 Mar 2025 22:12:25 GMT', 'Content-Length': '55'}) HTTP response body: b'{"code":"unauthorized","message":"unauthorized access"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/enpal/sensor.py", line 60, in async_setup_entry
    tables = await hass.async_add_executor_job(get_tables, config['enpal_host_ip'], config['enpal_host_port'], config['enpal_token'])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/enpal/sensor.py", line 31, in get_tables
    tables = query_api.query(query)
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/client/query_api.py", line 100, in query
    response = self._query_api.post_query(org=org, query=self._create_query(query, self.default_dialect, params),
                                          async_req=False, _preload_content=False, _return_http_data_only=False)
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/service/query_service.py", line 258, in post_query
    (data) = self.post_query_with_http_info(**kwargs)  # noqa: E501
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/service/query_service.py", line 338, in post_query_with_http_info
    return self.api_client.call_api(
           ~~~~~~~~~~~~~~~~~~~~~~~~^
        '/api/v2/query', 'POST',
        ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<12 lines>...
        collection_formats=collection_formats,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        urlopen_kw=urlopen_kw)
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/api_client.py", line 341, in call_api
    return self.__call_api(resource_path, method,
           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
                           path_params, query_params, header_params,
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
                           _return_http_data_only, collection_formats,
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                           _preload_content, _request_timeout, urlopen_kw)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/api_client.py", line 171, in __call_api
    response_data = self.request(
        method, url, query_params=query_params, headers=header_params,
        post_params=post_params, body=body,
        _preload_content=_preload_content,
        _request_timeout=_request_timeout, **urlopen_kw)
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/api_client.py", line 386, in request
    return self.rest_client.POST(url,
           ~~~~~~~~~~~~~~~~~~~~~^^^^^
                                 query_params=query_params,
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
                                 body=body,
                                 ^^^^^^^^^^
                                 **urlopen_kw)
                                 ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/rest.py", line 304, in POST
    return self.request("POST", url,
           ~~~~~~~~~~~~^^^^^^^^^^^^^
                        headers=headers,
                        ^^^^^^^^^^^^^^^^
    ...<4 lines>...
                        body=body,
                        ^^^^^^^^^^
                        **urlopen_kw)
                        ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/influxdb_client/rest.py", line 254, in request
    raise ApiException(http_resp=r)
influxdb_client.rest.ApiException: (401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.2.0', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Tue, 11 Mar 2025 19:56:35 GMT', 'Content-Length': '55'})
HTTP response body: b'{"code":"unauthorized","message":"unauthorized access"}'

and on the local access to [Solar Rel. 8.45.1-301788 (10.03.2025)](http://192.168.1.137/) there is no more /config menu option......

nopeee535, 2025-03-12:
Hi,
i had the same issue. I changed the following lines (106 - 108) in the sensor.py:

```
        elif measurement == "battery":
            if field == "Power.Battery.Charge.Discharge":
                addSensor('mdi:battery-charging', 'Enpal Battery Power', 'power', 'W')
```

adding : elif measurement == "battery":
and change the next line from elif to if solved it for me

btw.
some updates ago i ve lost the string power values -> i added this to the inverter section:

```
elif field == "Power.DC.String.1":
               addSensor('mdi:lightning-bolt', 'Enpal Power String 1', 'power', 'W')
            elif field == "Power.DC.String.2":
                addSensor('mdi:lightning-bolt', 'Enpal Power String 2', 'power', 'W')
            elif field == "Voltage.String.1":
                addSensor('mdi:lightning-bolt', 'Enpal Voltage String 1', 'power', 'V')
            elif field == "Voltage.String.2":
                addSensor('mdi:lightning-bolt', 'Enpal Voltage String 2', 'power', 'V')
            elif field == "Current.String.1":
                addSensor('mdi:lightning-bolt', 'Enpal Ampere String 1', 'power', 'A')
            elif field == "Current.String.2":
                addSensor('mdi:lightning-bolt', 'Enpal Ampere String 2', 'power', 'A')
```
hope that will help : )

fluidman, 2025-03-12:
Thanks the problem is, as shown in the logs that even though the integration seems to load fine it gets unauthorized errors and can't recreate the entities.

Also, when I load the IP of the inverter to reach the Solar Rep dashboard I don't have the /config page anymore....

fluidman, 2025-03-12:
What would I set as organization?

nopeee535, 2025-03-12:
looks good
are you sure your token is correct ?

defux-23, 2025-03-12:
> But since the config page is gone now, where would I find the current one now?

(Almost) same problem over here! We got our PV installed today and the interface (Solar Rel. 8.45.2-303733 (12.03.2025)) has no config page. Is there another way to get the token - otherwise I guess we'll have to write to Enpal for getting it?

NiclasPe, 2025-03-12:
@defux-23 @fluidman I faced the same Issue today and ended up scraping the data from the second Page of the Overview. So if you only nerd a few metrics, you could setup rest sensors for that. It’s maybe more reliable.

chrisbe58, 2025-03-13:
Same issue here with Solar Rel. 8.45.2-303733 (12.03.2025). Lost the battery information. Rest ist still working. Lucky me, I have not yet removed the integration and reinstalled it. So my DB key seems to be still valid and access is in principle granted.

chrisbe58, 2025-03-13:
> I reached out to Enpal's support and got a new working Token as well as a username and password with read access to the InfluxDB dashboard available on IP:8086
> 
> Now it's working and I would have to update the sensor.py as me too ed in previous comments

Battery values are available as well?

fluidman, 2025-03-13:
No they are not available, but i got it working using @nopeee535 fix above....

so you edit sensor.py, look for
`            elif field == "Power.Battery.Charge.Discharge":
                addSensor('mdi:battery-charging', 'Enpal Battery Power', 'power', 'W')`

and change it to 
`        elif measurement == "battery":
            if field == "Power.Battery.Charge.Discharge":
                addSensor('mdi:battery-charging', 'Enpal Battery Power', 'power', 'W')`

i also added below to the end of the inverter fields:
`            else:
                _LOGGER.debug(f"Not adding measurement: {measurement} field: {field}")`

plus i added a sensor for the battery temp together with the other battery sensors
`            elif field == "Temperature.Battery":
                 addSensor('mdi:thermometer', 'Enpal Battery Temperature', 'temperature', '°C')`

chrisbe58, 2025-03-13:
Ok cool. Will do it in a free timeslot the next days. Thanks for the infos and support!

chrisbe58, 2025-03-13:
Hm, had already today a free slot. Unfortunately it looks like the fix did not work for me. Is the token still the same or do I need to ask for a new one at Enpal support? Because all entities are not working any longer now...

fluidman, 2025-03-13:
I am guessing that the old token is just cached for everyone and will eventually expire for everyone that still has it active...

I opened a ticket yesterday at 5pm CET and got the credentials at 9:30am CET today.

chrisbe58, 2025-03-13:
That would explain the behavior. Ok, let me check with Enpal support

defux-23, 2025-04-03:
> [@defux-23](https://github.com/defux-23) [@fluidman](https://github.com/fluidman) I faced the same Issue today and ended up scraping the data from the second Page of the Overview. So if you only nerd a few metrics, you could setup rest sensors for that. It’s maybe more reliable.

Sure, that's an option, however I prefer the more sophisticated informations from the integration. And I guess Enpal will probably also change the solar rel interface from time to time in the future, so I guess overll getting stuff directly from the InfluxDB is better in the long run (hope Enpal does not decide to lock that somewhen in the future). I also got the token from Enpal customer service, though the gickowtf didn't provide all entities (c.f. the other open issues), but the TerranUlm fork works well for me:
https://github.com/TerranUlm/enpal-homeassistant/tree/TerranUlm-patch-1
There are still minor issues with the behaviour of some meters, but that's another issue! ;-)