gickowtf/enpal-homeassistant #30: Battery entity stopped working

eisbaer9, 2025-03-15:
Today the battery entity suddenly stopped working.
It says that the entity is no longer available.

eisbaer9, 2025-03-15:
There was an update on the enpal system:  Solar Rel. 8.45.2-303733 (12.03.2025)

Superflake, 2025-03-16:
Same here and the Webseite from enpal rel dont Show up all Menü enymore the Part off the config where the api toke Was...

maker-chrobok, 2025-03-16:
I have the same Problem wi the new Solar Rel. 8.45.2-303733 (12.03.2025) no Battery info, no config Link anymore

maker-chrobok, 2025-03-16:
The changes in [erran ulm patch 2 #27](https://github.com/gickowtf/enpal-homeassistant/pull/27) works on the new Solar Rel. 8.45.2-303733 (12.03.2025) Issue for me.

defux-23, 2025-03-21:
I just got my token from Enpal today, and the battery entities are not showing up as well - hope there's a way to fix it...

Superflake, 2025-03-25:
I hope so

defux-23, 2025-04-03:
I switched to the TerranUlm fork and there I can get all entities - works well for me now. There are still a few strange behaviours with some dupllicate entities and some entities having a daily "spike" to zero so one needs to be a bit careful in what entities to select for the energy card, but overall it's working ok:
https://github.com/TerranUlm/enpal-homeassistant/tree/TerranUlm-patch-1